oresat/oresat-c3-hardware #2: Solder mask pulled back too far

davidalay, 2020-08-14:
The solder mask is pulled back too far from the RF traces and some pads.  At some points it exposes other traces that should be covered by the solder mask.  Not a massive problem but should be fixed for the flight board.

glennl, 2020-10-16:
We determined that the solder mask T_Stop dimension on all new transmission lines with the name _***_PA$***_ can be changed. Also **_TX_TO_** and **_TX_OUT_** The new values can be changed to 3x the layer thickness (h). We find h=0.17mm so, 3*0.17 = 0.51mm. Since this is the dimension on each side, the total T_Stop would be 2x that value, or 1.02 mm. Currently the T_Stop is set to 2.1 mm.

andrewgreenberg, 2020-12-14:
Decided to move RF net clearance to 0.75 mm which is ~ 4x the dielectric thickness. Better than minimum of 3 and worse than the 5 some people say.

andrewgreenberg, 2020-12-14:
The TSTOP on the RF traces is 1.6 wide, which protects the ground copper with  0.18 mm of coverage on both sides.

andrewgreenberg, 2020-12-29:
Done! And in a markdown file in the doc folder.

closed 2020-12-29 by andrewgreenberg