oresat/oresat-c3-hardware #22: 5th order LPF between AX5043 SE output and PA needed

glennl, 2020-10-19:
The requirements document (block diagram) shows a 5th order LPF between the AX5043 SE output, and the input of the PA. This has not been added.

Glenn and Vigely should confirm it is really needed.

glennl, 2020-10-21:
After discussion with Vigely, it's determined it is needed. There are significant magnitude harmonic products coming off of the AX5043, and the PA design will likely amplify them. The filter choice is the one Vigely used in his thesis work. and can be found at:
**_oresat-c3-rf/eagle/lpf-5th-order.sch_**

NB: Since the signal is still considered "small signal" at this point before amplification, small 0402 lumped components can still be used.

There was also discussion about adding another 5th order LPF **after** the PA, but we decided to leave this out because, (1) the PA will not be running anywhere near its 1 dB compression point (low in its linear range), and (2) the component sizes will need to be bigger, and the board is already getting a bit full. After testing this C3 Rev-4 version, we will need to decide if the is a good decision.

andrewgreenberg, 2020-12-29:
Glenn, II need this designed. Can you do that?

glennl, 2021-01-03:
Working on this right now. Should be small 0402 components, and there is a bunch of space currently available on the board for this.

glennl, 2021-01-03:
Andrew, Thanks for placing the circuit. We can close this issue once I determine the lumped component values.

glennl, 2021-01-04:
The very same LPF is used for the UHF Tx between the AX5043 output and the PA input (436 MHz), and the L-Band's IF between the mixer output and the AX5043 input (452-462 MHz).

Checking the component values with a 5th order LC Chebychev LPF calculator, the values produce a filter of the following specs:
   **_Cutoff freq:  560 MHz
   Zo: 50 ohm
   Passband ripple: 1 dB_**

This has been confirmed with Vigely

closed 2021-01-04 by glennl