oresat/oresat-c3-hardware #35: U15 (Qorvo/Triquint 856930) is obsolete and isn't in stock anywhere

andrewgreenberg, 2021-03-08:
We have 8 left; we can do two more runs of the C3 and then we're screwed. Please choose a new filter ASAP. Bonus points if it's a direct replacement, but I won't hold my breath :)

glennl, 2021-03-08:
This is a 457.5 MHz SAW filter.

glennl, 2023-11-19:
This component is special and we need to treat the ones we have like gold. I have not yet found a reasonable replacement. We can move to another IF frequency allowing us to use another available part, however this will require us to change the LO (synthesizer) frequency. This will require us to do an analysis of the new image frequencies that are formed by this. Vigely did extensive work with this, so we will need to go back to his thesis to review his process. It's totally doable, but not trivial.