oresat/oresat-c3-hardware #47: Important: Add PCB inductance between Q7 emitter and ground

glennl, 2022-03-13:
There is an inductance formed by traces on the PCB between the two emitter pins of Q7, and the surrounding ground plane. This inductance is very small, but essential as a form of regeneration to maintain stability. The inductance was provided on the dev boards, but was changed as far back as v3. There was a fix applied to v5 boards by trimming away part of the PCB trace, and that confirmed its requirement, as shown in testing that the LNA breaks into oscillation when temperature is at or below room temp.

glennl, 2022-12-28:
This may be the same as issue #44?

andrewgreenberg, 2022-12-30:
Added emitter degenerattion like the "LEGO" board. Moving other components next door since we have more room now.

closed 2022-12-30 by andrewgreenberg