oresat/oresat-c3-hardware #48: Investigate AX5043 RX/TX matching circuit at UHF radio

zisi, 2022-10-12:
Hi, i would like to ask about the matching circuit in balanced and unbalanced lines in the AX5043. In both cases at 436.5MHz your design is different for the proposed design in [datasheet](https://www.onsemi.com/pdf/datasheet/ax5043-d.pdf). How you calculate these L-C values?
For example in TX the C65 and L139 and in RX the BaLun L11,C58,C59 and L10(+L12), how you calculate, which is the input/output impedance of AX5043 at ~430MHz?

Thanks

zisi, 2022-10-14:
According to [this](https://www.onsemi.com/forum/t/faq-how-to-operate-ax5043-at-27-mhz-custom-matching-networks-for-ax5043/274/7), i understand that you do not need the load network for internal differential PA, because you use the differential path only as RX, like this [application note](https://www.onsemi.com/download/application-notes/pdf/and9312-d.pdf). So you need only a BalUn, with input balanced impedance 100Ohm to unbalanced 50Ohm. I'm not sure about the usage of L12.

zisi, 2022-10-25:
Hi, there: https://gitlab.com/librespacefoundation/pq9ish/pq9ish-comms-vu-hw/-/issues/6#note_1141997003 you can find an analysis to reduce parts for RX and TX paths. I think we can close the issue.

Thanks to have your project open and share your knowledge about RF.

andrewgreenberg, 2023-08-29:
OK! Closing!

closed 2023-08-29 by andrewgreenberg