oresat/oresat-c3-hardware #52: Add specialized RF test points for easy probing

andrewgreenberg, 2023-03-01:
Come up with a common spacing for SMT pad (0.38 mm wide so it doesn't affect the trace) and a ground pin for both the RF probe and regular scope probes.

glennl, 2023-03-26:
At these test points point, it is expected that UT-047 semi rigid coax is to be attached. There is a pre-formed product from DigiKey ([931-1391-ND](https://www.digikey.com/en/products/detail/taoglas-limited/CAB-058/5287277?s=N4IgTCBcDaIJwGYCMBaJC6oHIBEQF0BfIA)) to implement this.

andrewgreenberg, 2023-08-29:
This is done; See new "mask only" TPs meant to be overlayed on the RF trace. There are two FAB circles for putting GND vias, and then the GND - signal - GND spacing of 1 mm nicely allows for probing with this test probe: https://www.ebay.com/itm/185381060789

closed 2023-08-29 by andrewgreenberg