oresat/oresat-c3-hardware #62: UHF TX 5th order LPF dropping 6 dBm

andrewgreenberg, 2023-10-25:
Either replace with values from 2023-10-24 testing (see #oresat-comms), or replace with some solid state device.

andrewgreenberg, 2023-11-13:
OK I've updated this to: L17, L18 are 18nH, and C87 = 3pF, C86 = 8.3 pF, and C85 = 3pF.

Close this issue if you want to stick with this, otherwise let me know better values and/or a good IC.

glennl, 2023-11-19:
Let;s keep this as is.
Rationale:
The components chosen gives a break frequency of around 700 MHz, which keeps the the passband S21 around -0.15 dB, and the first harmonic (which is what we want to supress) at 900 MHz with S21 around -10.5 dB. The insertion loss of 6 dB is the voltage loss, and power loss is 3 dB. This is expected, and SAW devices are not much better for overall S21. SAW's shine with steep skirts, but we are okay with this.

glennl, 2023-11-19:
Let's keep this as is.
Rationale:
The components chosen gives a break frequency of around 700 MHz, which keeps the the passband S21 around -0.15 dB, and the first harmonic (which is what we want to supress) at 900 MHz with S21 around -10.5 dB. The insertion loss of 6 dB is the voltage loss, and power loss is 3 dB, which is expected. SAW devices are not much better for overall S21. SAW's shine with steep skirts, but we are okay with this response.

closed 2023-11-19 by glennl